Xinyuan-LilyGO/T-Embed-CC1101 #2: De-brick instructions

tsbo, 2024-10-14:
I just bought my T-Embed CC1101 and managed to brick it using Arduino (maybe i did not provide any of the values from the screenshot, even though i checked them twice).
The device now is not lighting up and not showing any liveness at all, but erase flash and uploading are completed successful - without any visible changes.
Please provide details how to put it back to "normal"

ShallowGreen123, 2024-10-16:
你可以告诉我这些信息吗，为了帮助你解决问题。
1. 使用的什么环境编译的代码，是 PIO还是 Arduno？
2. 如果使用的是 Arduino，用的什么版本？
3. 是编译下载哪一个例程？

ShallowGreen123, 2024-10-16:
Can you tell me this information to help you solve the problem?
1. What environment did you use to compile the code, PIO or Arduno?
2. If you used Arduino, what version did you use?
3. Which example did you compile and download?

tsbo, 2024-10-17:
Hello, thank you for getting back to me!
I tried Arduino 1.8 and 2.2 for majority of the examples here (lvgl, display, cc1101). They are compiling, uploading - but then nothing is shown on the screen and device seems not working. I tried examples from T-Embed project - that are also compiling ,but not showing anything on the screen.
Following the instructions on sections Tips from https://github.com/Xinyuan-LilyGO/T-Embed , i managed to de-brick it. Trying with the ESP32 stock examples in Arduino, flashing is working, but reseting after flashing is not working - i need to unplug and plug back the usb cable, in order to restart the device. 
I needed to remove also the battery, in order to be able to restart the device using the usb cable

ShallowGreen123, 2024-10-17:
T-Embed-CC1101 and T-Embed are not compatible.

You can confirm whether your board is T-Embed-CC1101 or T-Embed.

Then, go to the corresponding Github repository and run the examples in it;

T-Embed-CC1101: https://github.com/Xinyuan-LilyGO/T-Embed-CC1101

T-Embed: https://github.com/Xinyuan-LilyGO/T-Embed

tsbo, 2024-10-17:
mine device is T-Embed CC1101 - the fresh version with 1300 mA battery

closed 2024-11-06 by tsbo