Xinyuan-LilyGO/T-Embed-CC1101 #3: LORA init failed

jiangxw, 2024-10-23:
If I go to Settings->About System, both "LORA init" and "TF Card" show "FAIL" on a brand new device. how do I fix that?

ShallowGreen123, 2024-10-23:
You can use `examples/CC1101_xxx` and `examples/tf_card_test` to test whether Lora and TF card are normal respectively;

jiangxw, 2024-10-23:
I flashed cc1101_send_irq to the device. But "LORA init" still shows up as "FAIL". In addition, the interface did not change. What did I miss?

The following is the log for flashing the firmware
------------------------------------------------------------------------------------
Retrieving maximum program size .pio/build/cc1101_send_irq/firmware.elf
Checking size .pio/build/cc1101_send_irq/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.1% (used 62584 bytes from 327680 bytes)
Flash: [==        ]  24.4% (used 1600073 bytes from 6553600 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbmodem2101
Uploading .pio/build/cc1101_send_irq/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem2101
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:ca:43:36:f0:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00196fff...
Compressed 15104 bytes to 10401...
Writing at 0x00000000... (100 %)
Wrote 15104 bytes (10401 compressed) at 0x00000000 in 0.2 seconds (effective 492.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 542.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 711.1 kbit/s)...
Hash of data verified.
Compressed 1600496 bytes to 1033142...
Writing at 0x00010000... (1 %)
Writing at 0x000171cc... (3 %)
Writing at 0x000251b3... (4 %)
Writing at 0x0002c8f7... (6 %)
Writing at 0x00031e47... (7 %)
Writing at 0x00037a9c... (9 %)
Writing at 0x00042e30... (10 %)
Writing at 0x0004e37d... (12 %)
Writing at 0x00059130... (14 %)
Writing at 0x0006d867... (15 %)
Writing at 0x00074ddf... (17 %)
Writing at 0x0007a3ca... (18 %)
Writing at 0x0007faaf... (20 %)
Writing at 0x000848ca... (21 %)
Writing at 0x00089c56... (23 %)
Writing at 0x0008ed1d... (25 %)
Writing at 0x00094610... (26 %)
Writing at 0x00099548... (28 %)
Writing at 0x0009e685... (29 %)
Writing at 0x000a389b... (31 %)
Writing at 0x000a9242... (32 %)
Writing at 0x000ae721... (34 %)
Writing at 0x000b3b8a... (35 %)
Writing at 0x000b878d... (37 %)
Writing at 0x000bd63c... (39 %)
Writing at 0x000c310f... (40 %)
Writing at 0x000c90d0... (42 %)
Writing at 0x000ced05... (43 %)
Writing at 0x000d4a2f... (45 %)
Writing at 0x000da786... (46 %)
Writing at 0x000dfd1b... (48 %)
Writing at 0x000e5973... (50 %)
Writing at 0x000eaaf3... (51 %)
Writing at 0x000efdf8... (53 %)
Writing at 0x000f4dac... (54 %)
Writing at 0x000f9cbd... (56 %)
Writing at 0x000fec62... (57 %)
Writing at 0x00103b27... (59 %)
Writing at 0x00108d31... (60 %)
Writing at 0x0010debf... (62 %)
Writing at 0x00112ef9... (64 %)
Writing at 0x00117d20... (65 %)
Writing at 0x0011cc3a... (67 %)
Writing at 0x0012265d... (68 %)
Writing at 0x001279d0... (70 %)
Writing at 0x0012cd8d... (71 %)
Writing at 0x00131b60... (73 %)
Writing at 0x00136adf... (75 %)
Writing at 0x0013bbef... (76 %)
Writing at 0x0014108b... (78 %)
Writing at 0x0014638c... (79 %)
Writing at 0x0014babe... (81 %)
Writing at 0x00151346... (82 %)
Writing at 0x00156a4c... (84 %)
Writing at 0x0015bf9f... (85 %)
Writing at 0x00161918... (87 %)
Writing at 0x001678fd... (89 %)
Writing at 0x00170219... (90 %)
Writing at 0x00177b3f... (92 %)
Writing at 0x0017d6ad... (93 %)
Writing at 0x0018569e... (95 %)
Writing at 0x0018affe... (96 %)
Writing at 0x0019058f... (98 %)
Writing at 0x00196629... (100 %)
Wrote 1600496 bytes (1033142 compressed) at 0x00010000 in 11.7 seconds (effective 1097.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=========================================== [SUCCESS] Took 24.01 seconds ===========================================

Environment      Status    Duration
---------------  --------  ------------
cc1101_send_irq  SUCCESS   00:00:24.013
============================================ 1 succeeded in 00:00:24.013 ============================================
 *  Terminal will be reused by tasks, press any key to close it.

ShallowGreen123, 2024-10-24:
Can I see the log output of the cc1101_send_irq log?

jiangxw, 2024-10-24:
The following is the output. Does it require a SD card to function? I used platformio to upload the firmware and it succeeded as in above log. But I'm not sure if the firmware actually take effect because nothing in the UI changed.

***************************************
[  1980][I][XPowersCommon.tpp:421] begin(): SDA:8 SCL:18
[  1980][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
getInputCurrentLimit: 3250 mA
getChargerConstantCurr: 832 mA
[CC1101] Initializing ... 315.00 MHz 
failed, code -16
[  2154][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
Found chip PN532
Firmware ver. 1.6
Waiting for an ISO14443A Card ...
EMBED_COLOR_BG:0x161823
EMBED_COLOR_FOCUS_ON:0xb4dc38
EMBED_COLOR_TEXT:0xffffff
EMBED_COLOR_BORDER:0xbbbbbb
EMBED_COLOR_PROMPT_BG:0xfffee6
EMBED_COLOR_PROMPT_TXT:0x1e1e00
[  2758][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2858][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2958][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3057][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  3058][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  3059][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3068][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3173][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3273][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3372][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card Mount Failed
[  3373][E][vfs_api.cpp:24] open(): File system is not mounted
Failed to open directory

ShallowGreen123, 2024-10-24:
You can try using the `factory_test` example. My tests are OK.

The `factory` example does have the problem you described, thanks for the feedback;

jiangxw, 2024-10-24:
I actually suspect that it's still running the default firmware rather than the firmware that I flashed. Anything other than flashing the firmware that I need to do?

jiangxw, 2024-10-24:
factory_test works sometimes, but it sometimes also fails to init cc1101 with error code -16.

I tried both cc1101_send_irq and cc1101_recv_irq, they both crash on boot. How do I fix this? Is there a good commit that I can try?

Here is the log for cc1101_send_irq:

ELF file SHA256: d5245029f2b9ee48

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376fd0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[    96][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[CC1101] Initializing ... 315.00 MHz 
failed, code -16

ShallowGreen123, 2024-10-24:
Consider replacing a CC1101 for you, it seems that there is a problem with Lora's SPI connection。

jiangxw, 2024-10-25:
What's the process to get it replaced?

jiangxw, 2024-10-27:
I turned on SPI debug and here is more log

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376fd0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[    96][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[CC1101] Initializing ... 315.00 MHz 
RLB_SPI: R      71      14
RLB_SPI: CMD    W       30      00
RLB_SPI: CMD    W       36      0F
RLB_SPI: R      75      0
RLB_SPI: R      75      1
RLB_SPI: R      18      F
RLB_SPI: W      18      1F
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: 
RLB_SPI: address:       0x18
RLB_SPI: bits:          5 4
RLB_SPI: value:         0x10
RLB_SPI: current:       0xF
RLB_SPI: mask:          0x30
RLB_SPI: new:           0x1F
RLB_SPI: read:          0x4
RLB_SPI: R      18      F
RLB_SPI: W      18      D
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: R      18      F
RLB_SPI: R      18      FF
RLB_SPI: R      18      4
RLB_SPI: 
RLB_SPI: address:       0x18
RLB_SPI: bits:          1 1
RLB_SPI: value:         0x0
RLB_SPI: current:       0xF
RLB_SPI: mask:          0x2
RLB_SPI: new:           0xD
RLB_SPI: read:          0x4
failed, code -16
E (10447) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10447) task_wdt:  - IDLE (CPU 0)
E (10447) task_wdt: Tasks currently running:
E (10447) task_wdt: CPU 0: loopTask
E (10447) task_wdt: CPU 1: IDLE
E (10447) task_wdt: Aborting.

abort() was called at PC 0x4200b34c on core 0

Backtrace: 0x403773c2:0x3fc936a0 0x4037ab81:0x3fc936c0 0x403804f9:0x3fc936e0 0x4200b34c:0x3fc93760 0x40378529:0x3fc93780 0x42001add:0x3fcebeb0 0x4200593a:0x3fcebee0

  #0  0x403773c2:0x3fc936a0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037ab81:0x3fc936c0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x403804f9:0x3fc936e0 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4200b34c:0x3fc93760 in task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/task_wdt.c:176 (discriminator 3)
  #4  0x40378529:0x3fc93780 in _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_vectors.S:1114
  #5  0x42001add:0x3fcebeb0 in setup() at examples/cc1101_send_irq/cc1101_send_irq.ino:79
  #6  0x4200593a:0x3fcebee0 in loopTask(void*) at /Users/xiaowei/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 2653754c38836ee4

closed 2024-12-16 by github-actions[bot]