Xinyuan-LilyGO/T-Embed-CC1101 #4: Can't provision WiFi using EspTouch

jiangxw, 2024-10-23:
I tried to configure WiFi and got a timeout. The "help" tells me to configure using EspTouch. So I followed, but the device doesn't show up in either EspTouch or EspTouch V2. How can I configure WiFi?

ShallowGreen123, 2024-10-23:
1. To download the EspTouch software, you can scan the QR code below to download;
[image]

2. Enter EspTouch on your phone. Please note that your phone must be connected to WIIF first (not 5G), then click EspTouch.
[image]

3. Enter the WIFI password on your phone; the device should be in the WIFI interface;
[image]

4. Click Confirm on your phone; click Config on your device; then wait for the network configuration to succeed;
[image]

5. The following is a successful network configuration
[image]

jiangxw, 2024-10-23:
That worked. thanks!

closed 2024-12-07 by github-actions[bot]

tally0highwayman, 2025-01-28:
I followed these exact steps and they did not work on my new, out-of-the-box T-Embed C1101.
SF Version: v1.1-20250109
HD Version: v1.0-20240729