Xinyuan-LilyGO/T-Embed-CC1101 #9: Battery charge

bmorcelli, 2024-11-06:
Hello, 

I can´t get the battery to charge when connected to USB, unless I get back to Factory app

I'm using this code:
lib: lewisxhe/XPowersLib @ ^0.2.6

```
#include <XPowersLib.h>
extern XPowersPPM PPM;

void setup() {
// code code

Wire.begin(8, 18);
      pmu_ret = PPM.init(Wire, 8, 18, BQ25896_SLAVE_ADDRESS);
      if(pmu_ret) {
          PPM.setSysPowerDownVoltage(3300);
          PPM.setInputCurrentLimit(3250);
          Serial.printf("getInputCurrentLimit: %d mA\n",PPM.getInputCurrentLimit());
          PPM.disableCurrentLimitPin();
          PPM.setChargeTargetVoltage(4208);
          PPM.setPrechargeCurr(64);
          PPM.setChargerConstantCurr(832);
          PPM.getChargerConstantCurr();
          Serial.printf("getChargerConstantCurr: %d mA\n",PPM.getChargerConstantCurr());
          PPM.enableADCMeasure();
          PPM.enableCharge();
      }
// code code 
}

```

ShallowGreen123, 2024-11-07:
Try `examples/bq25896_test`

The example uses the library lewisxhe/ XPowersLib@ \^0.2.4

After downloading this program, open the serial port;

bmorcelli, 2024-11-07:
I changed the library and tested with the example setup, didn´t work either..

it only worked afer add:
´´´
     PPM.enableOTG();
     PPM.disableOTG();
´´´

the working code was:

```
    bool pmu_ret = false;
    Wire.begin(8, 18);
    pmu_ret = PPM.init(Wire, 8, 18, BQ25896_SLAVE_ADDRESS);
    if(pmu_ret) {
        PPM.setSysPowerDownVoltage(3300);
        PPM.setInputCurrentLimit(3250);
        Serial.printf("getInputCurrentLimit: %d mA\n",PPM.getInputCurrentLimit());
        PPM.disableCurrentLimitPin();
        PPM.setChargeTargetVoltage(4208);
        PPM.setPrechargeCurr(64);
        PPM.setChargerConstantCurr(832);
        PPM.getChargerConstantCurr();
        Serial.printf("getChargerConstantCurr: %d mA\n",PPM.getChargerConstantCurr());
        PPM.enableADCMeasure();
        PPM.enableCharge();
    }
     PPM.enableOTG();
     PPM.disableOTG();
```

thank you for your help.

closed 2024-11-07 by bmorcelli