Xinyuan-LilyGO/T-Embed-CC1101 #15: power drain

GuMAZ777, 2025-02-05:
how do you full turn off embed cc1101. i tried pressing the button on the inside to no avail. even holding it down for 18sec. all it did then was shut down and start back up. i’m running the bruce firmware.  couldn’t they add full shut down to config?. because all that does is put it into a deep sleep mode. thanks

ShallowGreen123, 2025-02-06:
The PN532 supports operating at 13.56 MHz. The official description of the PN532 is incorrect.

ShallowGreen123, 2025-02-06:
To fully disable T-Embed-CC1101, you need to disable the BQ25896 power chip;

Long press the PWR button in T-Embed-CC1101 to start the BQ25896 chip power supply;

Refer to closing BQ25896 [examples/bq25896_shutdown](https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/tree/master/examples/bq25896_shutdown)

GuMAZ777, 2025-02-06:
tried it with the battery connected. still nothing. is there a video on it?

ShallowGreen123, 2025-02-06:
I just modified the program to facilitate testing;

1. First download the program [examples/bq25896_shutdown](https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/tree/master/examples/bq25896_shutdown); The function of this program is to shut down after 5S.

2. Then, when using battery power, without inserting USB, measure the voltage according to the following figure.
[image]

3. Then long press the PWR key to power on (there will be voltage on the ammeter), and automatically shut down after 5 seconds (the ammeter voltage is 0); After shutdown, press PWR to start up and reciprocate this process;

The above tests the T-Embed-CC1101 procedure;

closed 2025-03-23 by github-actions[bot]