Xinyuan-LilyGO/T-Embed-CC1101 #28: Nfc does not work

K0IN, 2025-04-23:
When running the example code for emulating a tag, i always run into this issue:

```
[  3970][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[...]
```

https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/blob/master/examples/pn532_emulatetag/pn532_emulatetag.ino

run4type, 2025-05-11:
Did you find a way to emulate?

K0IN, 2025-05-21:
> Did you find a way to emulate?

no, do you have the same error?  (I was thinking maybe my hardware is defect)

Huzzla101, 2025-07-25:
So what kind of SW are you using to try to emulate, you also have a
chameleon etc that you connect to ? Are you sure that the pinouts are
properly connected according to the SW you using ?

closed 2025-09-10 by github-actions[bot]