Xinyuan-LilyGO/T-Embed-CC1101 #29: T-embeds wheel issue

Kotok7, 2025-04-26:
I have t-embed cc1101 for over 2 months now, but suddenly my wheel started to scroll things by itself in random directions. And it's not a firmware bug, because I tested 4 firmwares and nothing changed, so it's a hardware issue. Here's the video where that issue is clearly visible on official firmware. Please tell me what I need to do to repair this. https://github.com/user-attachments/assets/7fdf52ea-d06c-4676-9147-c2798c64fb54

IgnasFalcon, 2025-05-08:
I had a similar problem.

Kotok7, 2025-05-08:
I just cleaned it and it works, nevermind

closed 2025-05-08 by Kotok7

IgnasFalcon, 2025-05-08:
Cool! How did you clean it though?

Kotok7, 2025-05-09:
I just completely took the wheel apart, blew on it to get rid of the dust, and put it back together.

IgnasFalcon, 2025-05-13:
How did you take it of? Can you give a link to a video that shows taking them off an encoder?

Perelol, 2025-05-15:
Just pull it off clean using Isopropyl alcohol and re assemble pulling heads dome force don't lose black piece inside it just take it off face facing up

Kotok7, 2025-05-17:
> How did you take it of? Can you give a link to a video that shows taking them off an encoder?

I don't have any video. I just unscrewed everything and cleaned it, it's very easy to do