Xinyuan-LilyGO/T-Embed-CC1101 #30: Adding an NRF24 module and 2 external antennas on a custom 3D printed cover.

Mawa000, 2025-05-12:
I have installed an NRF GT-24 module on the T-Embed. Additionally, I have created a back cover to accommodate two external antennas in order to avoid drilling into the original case and to keep the aesthetics cleaner.

I have followed the assembly tutorial from this Reddit thread, which includes a video on YouTube. NOTE! There is a modification in the power connection, which is now done through the CC1101 and not through the ESP32.
https://www.reddit.com/r/LilyGO/comments/1k4u1ly/lilygo_tembed_nrf24_wiring/?share_id=GwPeiDFau2X5DQ9BmgpRF&utm_content=1&utm_medium=android_app&utm_name=androidcss&utm_source=share&utm_term=1

In the following link you can find the print files to do it yourselves:
https://www.printables.com/model/1293527-lilygo-t-embed-back-cover-for-two-antennas

Below I leave a few images as a guide of my installation process. In my case, I decided to use the same antenna for the original board's WIFI connection and the 2.4Ghz module while working on the same frequency. To do this, I used an SMA connector => 2 IPX.

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mawa000, 2025-05-14:
I finally changed the switch to use the IPX connector instead of the PCB antenna and it has improved the range. However, the coverage is still limited, but I suppose it is due to a hardware limitation.
I consider the issue resolved. In the future, I will try to install a higher power one like the E01-2G4M27D. I will keep you updated on the progress.

closed 2025-05-14 by Mawa000

Mawa000, 2025-06-27:
> > Finalmente cambié el interruptor para usar el conector IPX en lugar de la antena PCB y ha mejorado el alcance. Sin embargo, la cobertura sigue siendo limitada, pero supongo que se debe a una limitación de hardware. Doy por resuelto el problema. En el futuro, intentaré instalar uno de mayor potencia como el E01-2G4M27D. Os mantendré informados sobre el progreso.
> 
> ¿Se puede enviar una foto de cómo está GT24 después de cambiar para usar la antena externa?

I don't understand you. Do you mean how to connect the antennas to each connector?

breadDOA, 2025-06-27:
I didn't understand how to enable the outer antenna, have to do a jumper in 1 and 3? Is there a photo?

Mawa000, 2025-06-27:
> I didn't understand how to enable the outer antenna, have to do a jumper in 1 and 3? Is there a photo?

Ahhh! Okay, now I understand!
Yes. There are three connection points, 1, 2, ad 3. By default, it comes with an element (I don't know if it's a coil or a simple jumper) soldered between points 1 and 2. You must remove it and re-solder it between points 1 and 3. Point 1 is the center.

breadDOA, 2025-06-28:
But to weld in 1 and 3 I would need a "jumper" bigger than what comes in it, didn't you? Honestly I don't know how to do this

Mawa000, 2025-06-28:
> But to weld in 1 and 3 I would need a "jumper" bigger than what comes in it, didn't you? Honestly I don't know how to do this

Not in my case. The element is very small, but two drops of tin should be enough. I did it this way. But as I said, you have to be very patient and steady, because the element is very small. If you lose it for whatever reason, try making a tin bridge. If it's just a jumper, it should work.
You must move the jumper in the image to the position indicated by the arrows. The distance between the center and outer contacts is the same.

[image]

breadDOA, 2025-06-28:
The "metal" of the photo has to be expanded to pass over the contact 2 to be between 1 and 3 but not 2? Or just pass the estio in all 3, I'm not sure how to do this, you did not do it in yours?

breadDOA, 2025-06-28:
Can you record a video explaining or showing how to do it?

Mawa000, 2025-06-28:
> Can you record a video explaining or showing how to do it?

Sorry, but I think I can make a video.
It's simple. Right now, there's an SMD component (the one below the arrows I drew) that connects the center point and the point that goes to the PCB antenna (the copper track that makes a loop). You must remove it and re-solder it between the center point and the point I indicated with the green arrow, on the opposite side of the current one. The point on the PCB antenna board should not be connected anywhere.
Technically, it's a physical switch, connecting the center point to one end or the other. When you're done, if you've done it correctly, there should be electrical continuity between that component and the center pin of the IPX connector, but not with the PCB antenna. If you're not entirely sure what I'm explaining, I think it would be advisable to have someone with more experience do the installation for you.

geoloxy, 2026-10-01:
Странный символ не соответствует официальному трактату.

[image]