Xinyuan-LilyGO/T-Embed-CC1101 #31: LOW RANGE WIFI

peppemas1, 2025-05-15:
hi, i put an external antenna on my t-embed cc1101 and all work fine with firmware bruce, but now i have noticed problem on wifi strenght, i have try to change antenna, reconnect the original , change firmware(capibabara and original fw), but nothing, signal streght is decreased. How i can do ?

peppemas1, 2025-05-16:
Please answer, i can try this guide by espressif for factory reset esp32:

https://www.youtube.com/watch?v=3oEvXhgHZHo&t=198s

ShallowGreen123, 2025-05-16:
Adjust the transmission power: Set the power to the maximum value (such as 20dBm) through `esp_wifi_set_max_tx_power()`;

The general method to increase the receiving power is to replace the antenna with a better performance one. The surface mount antenna used in this product may not have as good performance parameters as the rubber rod antenna.

peppemas1, 2025-05-16:
the original antenna was replaced with external antenna mod ( i use a 3dbi antenna) , everything was working fine until a few days ago. But for a few days i have noticed low signal in receiver mode, i use bruce firmware , when do a wifi network scan i found only my  network with poor signal -70 near the modem ( 1 meter distance) , before in the same spot i found other wifi network near me, if i do the search by a smarphone or a m5stick i found the other network . If i use evil portal in bruce firmware , work perfect at long distance. The problem is only in wifi receiver mode

peppemas1, 2025-05-19:
> gh `esp_wifi_set_max_tx_power()`;

hi, how i can send this command to esp32, is possible via web

ShallowGreen123, 2025-05-20:
Hello, you can contact the customer service and ask them to send you a new WIFI antenna；

closed 2025-05-20 by peppemas1