Xinyuan-LilyGO/T-Embed-CC1101 #36: The backlight is on, but no content is shown on the display.

hoanb1, 2025-06-17:
Dear LilyGO Support Team,

I recently purchased a new LilyGO T-Embed CC1101 device through an online individual seller. While the device appears to be functioning in most aspects—buttons are responsive and I can see output through the serial monitor—the screen is not displaying anything. The backlight is on, but no content is shown on the display.

[image]
[image]

I have not modified the hardware, and all other components seem to be working normally. I would appreciate your assistance in diagnosing and resolving this issue.

Thank you in advance for your support.

ShallowGreen123, 2025-06-18:
1. You can try the individual screen test program first，[display_test](https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/tree/master/examples/display_test)

2. It might be due to poor screen contact. You can try disassembling the device and reinserting the screen's ribbon cable to see if it works.

3. If the screen still has problems after operating in the above way, you need to contact the merchant to replace it.

Huzzla101, 2025-06-18:
First try to use a toothpick and push the RST (reset) button that is on the
upper left side of battery , u dont have to remove battery but make sure u
get a tick when pushing the button... That will reset and may reboot screen
to work... IF that dont work then try to hold the POWER button on the upper
right side of the battery and hold it down for 7seconds then release then
push reset again and see if anything, Then you should go for the solution
from Shallowgreen after if it doesnt work....

JudgeLP, 2025-07-18:
You are not the only one with this problem, and Lilygo Official Store at AliExpress is being informed about it: https://github.com/pr3y/Bruce/issues/1455

closed 2025-09-02 by github-actions[bot]